git-afsantos/haros #1: export_manager.py: AttributeError: 'FileRuleViolation' object has no attribute 'metric'

Minipada, 2016-12-21:
Hello,
First thing, amazing work!
I was testing haros and got an issue
`AttributeError: 'FileRuleViolation' object has no attribute 'metric'` pointing to export_manager.py

I don't exactly know what it is, how it is working basically, but I commented out lines **196-197** and it worked.

I hope it can help track a bug.

Regards

git-afsantos, 2017-01-24:
Hi there!
Thank you for your feedback, and sorry for the late response. I've been absent from GitHub, and for some reason I didn't receive an email regarding your issue.

I'm also sorry that I haven't put up some manual or user instructions yet, but I will try to do so in the near future.

Indeed, you detected a _copy-paste_ bug that went unnoticed. I think it should work properly now (I'm still testing this tool).

Let me know how your experience went and what you were trying to do. I've submitted a few more commits since you opened this issue, and all feedback is welcome.

Regards

closed 2017-01-24 by git-afsantos

Minipada, 2017-01-25:
Great, nice job. I'll try it again very soon.
Just a tip, if you commit and add "Fixes #1" (#1 being the number of the issue), it will automatically close the issue and link it to the commit

git-afsantos, 2017-01-25:
Thanks for the tip. Will do that from now on.