git-afsantos/haros #17: Add screenshot to README

davetcoleman, 2017-11-15:
Showing the output of the visualizer will help new users like myself quickly understand what the output is. 

Also, from reviewing the README its not clear what kind of errors/suggestions the static analyzer will find/make.

git-afsantos, 2017-11-15:
Hello, Dave!

That is a good idea. There is a [demo of the visualizer](https://git-afsantos.github.io/haros/), but I haven't explicitly put a link anywhere, so I understand that it may be somewhat pointless. I will prepare a couple of screenshots and describe what is being shown.

Regarding the error reports, it is not a single static analyzer with a fix set of rules, so it really depends on what plugins you are running. I shall probably write a separate document about what each plugin does, and then provide a link to that.

git-afsantos, 2017-11-15:
Added a few screenshots in https://github.com/git-afsantos/haros/commit/f8a6ac1433560338ad1e62e053f888f773e6338f to illustrate the main views of the visualizer.

davetcoleman, 2017-11-17:
Awesome thanks!

closed 2017-11-17 by davetcoleman