git-afsantos/haros #20: Enhancement: Let the user know about loaded plugins

fkromer, 2017-11-28:
I think it would make sense to let users know about which plugins have been loaded (even better: before the full analysis is run at all). Maybe as part of the command line output ([`[HAROS] Loading plugins...`](https://github.com/git-afsantos/haros/blob/9109dbd4c9fe7e74c00f03f27eae0ca25c1a7284/haros/haros.py#L456)). This info is required for proper use of black- and white-listing. If it is not logged already the info

git-afsantos, 2017-11-29:
I agree. The only way for an user to check which plugins are loaded right now is to look at the plugins that come with the repository -- not practical. One of my to-dos is to rework the way plugins are installed. Haros would come without plugins, and the user would install them selectively.

```bash
haros install PLUGIN_NAME
```

This approach also minimises the dependencies of Haros. Otherwise, they will only keep increasing.

Furthermore, when I get the chance to implement a *preferences* file, it would be nice to allow users to enable/disable plugins right there, instead of requiring long shell commands.

```yaml
plugins:
    one_plugin:
        args: pass custom args here
    another_plugin:
        enabled: false
```

fkromer, 2017-11-29:
> Haros would come without plugins, and the user would install them selectively.
> There could be some pre-defined plugin packs, to avoid installing a lot of plugins individually.
> This approach also minimises the dependencies of Haros. Otherwise, they will only keep increasing.

Minimizing dependencies is great.

> Furthermore, when I get the chance to implement a preferences file, it would be nice to allow users to enable/disable plugins right there, instead of requiring long shell commands.

That would be great as well...

gavanderhoorn, 2017-11-29:
@git-afsantos wrote:
> Haros would come without plugins, and the user would install them selectively.

may I suggest to try to avoid growing Haros into a pkg manager for plugins? Where would the plugins be installed to? Would it still use `pip`? In that case a `virtualenv` makes a lot of sense.

I'm currently working on packaging haros for distribution (and use) in a binary/compiled ROS environment, and I'm not sure having Haros install plugins on its own is making that easier.

git-afsantos, 2017-11-29:
@gavanderhoorn :
> may I suggest to try to avoid growing Haros into a pkg manager for plugins?

That is absolutely a battle I am not willing to fight. Making a decent package manager is too much work (especially for one dev. alone), and, besides, that is an already solved problem. There are good tools for that out there.

What I meant, although I did not explain it, was that Haros would function pretty much as it does now, i.e., download the `haros_plugins` repository into the default `.haros/plugins` directory. The main differences would be:

1. `setup.py` in `haros` would not list external dependencies it has, that only serve for specific plugins;
2. `plugin.yaml` within each plugin would list external dependencies.

If a plugin does not have any dependencies (all functionality is implemented within `plugin.py`), it is ready to go when it comes with Haros. If a plugin has external dependencies, it would be skipped until the user runs `haros install <plugin>`. This command, in turn, could automate part of the installation (e.g. if the dependencies come from `pip`), or, otherwise, tell the user what to install manually.

> `haros install cppcheck_plugin`
> [HAROS] cppcheck_plugin requires external dependencies:
>   - cppcheck
> 
> Please run `apt-get install cppcheck` so the plugin can be installed.

In hindsight, this is not exactly *installing* plugins. It is as if the plugins came disabled by default and you had to enable which ones you want after installing dependencies. But I can see that even this idea would probably make your work quite harder. Do you think it best to scrap this idea, and just go for a cleaner way to let users know which plugins are loaded (e.g. the preferences file above)?

git-afsantos, 2018-03-08:
The new version of HAROS, as of #22 , displays the lsit of loaded plugins to the user.

Users can now also specify a default blacklist in a configuration file.

closed 2018-03-09 by git-afsantos