git-afsantos/haros #21: Allow analyse -d to create a new folder

ipa-jfh, 2017-12-19:
I would like to use a not-yet-existing folder for the given directory of `cmd analyse -d` and use the functionality of `viz.install()` to create it.
This is prevented by the following cmd:
https://github.com/git-afsantos/haros/blob/9109dbd4c9fe7e74c00f03f27eae0ca25c1a7284/haros/haros.py#L158
@git-afsantos is it possible to remove this line or can you suggest a way to support this feature? 

For reference:
> haros analyse -d DATA_DIR
> 
> Export analysis results to the given directory, instead of the default one. This option will also install the visualisation files. If DATA_DIR contains a previous analysis database for the current project within its tree, it will be loaded and new results will be added to that database.

git-afsantos, 2017-12-19:
Well, it depends on how you are running it. If it is a manual run, or as part of some script, you can just add a line to the script that creates the required directory.

```bash
mkdir -r DATA_DIR
haros analyse -d DATA_DIR
```

If it is more complicated than that, I can change the behaviour, and perhaps add a warning.

ipa-jfh, 2017-12-20:
Yes I agree this is a good way. I will close the issues

closed 2017-12-20 by ipa-jfh