git-afsantos/haros #23: Use under Windows?

gavanderhoorn, 2018-03-09:
More curious than a serious need, but is there a possibility to use Haros under Windows?

Provided dependencies (like `cppcheck`) can be run there, are you aware of any blockers in Haros itself?

gavanderhoorn, 2018-03-09:
Congrats on the v3 release btw. Very nice work :hamburger:

git-afsantos, 2018-03-09:
Thanks!

I am not sure Haros can work on Windows without implementing a lot of additional compatibility glue. The visualiser, at least, is obviously platform-independent, and already works on Windows. The problem is the analyser. The most certain answer is that it *may* work, with varying degrees of functionality.

Skimming through the scripts, I believe the basic analysis and file export procedures might work without much trouble. How much of the basic analysis works depends, of course, on the selection of enabled plugins.

Even before the analysis takes place, it will run into trouble when searching for packages, nodes and such. This feature depends on `rospkg` (Python library), and I am not sure it is available for Windows. Either one must implement a replacement (maybe a naïve one, that searches only in a hard-coded directory), or I might have to change something else in the settings (maybe allow the user to provide package-path pairs during initialization). For nodes, it needs the location of the ROS workspace, but I think it might be possible to get around this.

Another big problem is the most recent file parsing feature. Parsing of launch files is using `rosgraph` (again, this is not a problem to replace with a dumb implementation, I did it once in the early stages), and parsing of C++ files is using `libclang` from the Clang compiler. There might be a workaround for Clang...

I am doing development across multiple machines (Windows and multiple variants of Ubuntu for ROS Indigo and Kinetic). One of the machines I use the most is an older ROS Indigo, which only has an old version of `cppcheck`. I am aware that newer versions are able to do fuzzy parsing of C++ and dump the parse trees into XML files. I just did not have the time, but if I ever get my hands on it, this might be the alternative parser I provide, instead of Clang.

gavanderhoorn, 2018-03-09:
re: `rospkg` and `rosgraph`: at least the former works under Windows. I use it there as well. The latter probably does too, but might need some tweaking.

re: libclang: yeah, that's a bit more uncertain for me.

git-afsantos, 2018-03-09:
Just for reference, I think the minimum required version of `cppcheck` is [**1.66**](https://sourceforge.net/p/cppcheck/news/2014/08/cppcheck-166/) (2014-08-02).

> Generate xml dump of AST/ValueFlow/SymbolDatabase/TokenList if `--dump` is given