git-afsantos/haros #26: Add Python 3 support

git-afsantos, 2018-10-27:


asmodehn, 2019-06-17:
Hi, I m just dropping a note here to give a pointer to https://wily.readthedocs.io/en/latest/

git-afsantos, 2019-06-19:
Thanks for the pointer @asmodehn. I was not aware of this tool yet.

max-krichenbauer, 2019-09-02:
I ran a quick update of the current HAROS version from to Python3 to see if I would run into any problems:
https://github.com/max-krichenbauer/haros/tree/python3
I did not have any issues - everything seems to be running smoothly.

git-afsantos, 2019-09-04:
@max-krichenbauer Thanks for trying it out! I figured most of the changes would be due to the iterators and some imports, but I am glad that there is some confirmation now.

mikaelarguedas, 2020-11-23:
I'm interested in using this package in ROS Noetic (Python3 only). @git-afsantos is Python3 support something that is in the roadmap of haros ? do you foresee any blocker for this moving forward? I'm not familiar with haros but can maybe give a hand in the porting/testing if it's something that you're interested in supporting/releasing in Noetic.

git-afsantos, 2020-11-24:
@mikaelarguedas Python3 is definitely going to be added sometime in the future (soon™).
As you can see from the comment and preliminary experiments of Max, above, it seems that most of HAROS is already compatible.

One of the pending goals is to add support for ROS2, which also requires Python3, so I will have to do it.
I am just getting done with the last stages of my thesis, and afterwards I will be focusing more on the tool.

git-afsantos, 2020-11-27:
@mikaelarguedas I have gone through the code of HAROS, [bonsai](https://github.com/git-afsantos/bonsai), and some HAROS plugins ([1](https://github.com/git-afsantos/haros_plugins), [2](https://github.com/git-afsantos/haros-plugin-model-ged), [3](https://github.com/git-afsantos/haros-plugin-pbt-gen), [4](https://github.com/git-afsantos/haros-plugin-pyflwor)) to try and make them compatible with Python 3. After running a few preliminary tests, I think most of the issues should be gone.

In any case, if you are willing, you could try out running HAROS from source, and let me know if you run into any problems on your machine. Be sure to check out the [`dev-py3-support` branch](https://github.com/git-afsantos/haros/tree/dev-py3-support).

If everything goes well, I should be able to open a PR and make a release for this version.

mikaelarguedas, 2020-11-27:
That's awesome thanks! I'll give it a try over the weekend

mikaelarguedas, 2020-12-21:
@git-afsantos I could get a python3 version running based on the `dev-py3-support` with a couple changes. The diff can be seen at https://github.com/git-afsantos/haros/compare/dev-py3-support...mikaelarguedas:dev-py3-support?expand=1
The main changes are making the dockerfile work on ROS Noetic and Focal system dependencies as well as fixing crashes due to missing arguments in the full parser

git-afsantos, 2020-12-30:
Thanks for your contribution, @mikaelarguedas ! :+1:

ipa-hsd, 2021-03-26:
I needed some small changes for @max-krichenbauer's `python3` branch to work: https://github.com/max-krichenbauer/haros/pull/1. I have not tested https://github.com/git-afsantos/haros/tree/dev-py3-support yet.

Should `pip3 install haros` install one of these 2 branches?

git-afsantos, 2021-03-26:
No, @ipa-hsd.
These branches were not released yet, it must be cloned from the source. I should probably include the changes in the next release, though.

git-afsantos, 2021-08-06:
The changes to support Python 3 have been merged in #109, and released as [HAROS v3.10.4](https://github.com/git-afsantos/haros/releases/tag/v3.10.4) .
If there are further problems or incompatibilities, please report them in a new issue,

closed 2021-08-06 by git-afsantos