git-afsantos/haros #29: No module named bonsai.model

droter, 2018-11-24:
Following the Method 1 install.  I am getting this error:

File "./haros-runner.py", line 24, in <module>
    from haros.haros import main
  File "/home/matt/lawn_tractor_ws/src/haros/haros/haros.py", line 102, in <module>
    from .extractor import ProjectExtractor
  File "/home/matt/lawn_tractor_ws/src/haros/haros/extractor.py", line 34, in <module>
    from bonsai.model import CodeGlobalScope, pretty_str
ImportError: No module named bonsai.model

I have installed the dependencies listed on the README.  Is Bonsai this a separate install?

Thanks for your help,

Matt

droter, 2018-11-24:
Update:  Method 2 installed bonsai

git-afsantos, 2018-11-24:
It is a dependency that I forgot to add. Thanks for reminding me, I will fix both the instructions and the dependency list.

In the meantime:

```bash
pip install bonsai-code
```

droter, 2018-11-24:
Thanks Andre

commit a62d3b44 (2018-11-24): fixes #29 and fixes #30
closed 2018-11-24 by git-afsantos