git-afsantos/haros #31: HAROSviz

droter, 2018-11-24:
$ ./haros-runner.py analyse -p my_project.yaml -d temp -b cccc_plugin
[HAROS] Loading common definitions...
[HAROS] Loading plugins...
  > Loaded mi_calculator
  > Loaded cpplint_plugin
  > Loaded cppcheck_plugin
  > Loaded pylint_plugin
  > Loaded lizard_plugin
  > Loaded ccd_plugin
  > Loaded radon_plugin
[HAROS] Reading project and indexing source code...
[HAROS] Running analysis...
[HAROS] Saving analysis results...

Looks like it ran okay.  On the HAROSviz page it just shows "Loading...." 

Is there some where to check to see why it isn't loading the json file?

Thanks,

Matt

git-afsantos, 2018-11-24:
It could be a number of reasons. Given that command, I am assuming you are trying to open the `index.html` within `temp` in your browser.

If so, it could be that your browser is not set up with the permissions to open local files (without a server). This is often the case with Chrome.

If you open the browser's console, we might get a better idea of the error.
*Ctrl + Shift + J* (Windows / Linux) or *Cmd + Opt + J* (Mac) does the trick, or you can access it via the menus.

git-afsantos, 2018-11-24:
Alternatively, try opening with the `viz` command, and see if that works.

https://github.com/git-afsantos/haros#haros-viz--d-data_dir

droter, 2018-11-24:
Yes, that worked to run it directly.  Probably a local permissions issue like you mentioned.

Thanks for your help.

closed 2018-11-24 by droter