git-afsantos/haros #32: ROS2 Support

droter, 2018-12-09:
I was running HAROS against the latest release and there are a few errors on the packages:
https://raw.githubusercontent.com/ros2/rosdistro/ros2/crystal/distribution.yaml

https://github.com/ros2/rmw_implementation.git 
File "/usr/local/lib/python2.7/dist-packages/haros/cmake_parser.py", line 530, in _process_link_libraries
    assert len(args) >= 2
AssertionError

https://github.com/ros2/orocos_kinematics_dynamics.git
AttributeError: 'list' object has no attribute 'replace'

Some of the ROS2 meta packages don't contain a CMakeList.txt so they all fail.
https://github.com/ros2/ros2cli.git
IOError: [Errno 2] No such file or directory: '/home/matt/haros/ros/test_dir/ros2cli/CMakeLists.txt

Will ROS2 tests be different?

Thanks,

Matt

git-afsantos, 2018-12-09:
HAROS was not made with ROS2 in mind. If there are any file structure changes, such as missing CMake files, it will not work.

I will look into the first two errors, though, to see if there is some quick fix for them.

git-afsantos, 2018-12-17:
> https://github.com/ros2/rmw_implementation.git
> File "/usr/local/lib/python2.7/dist-packages/haros/cmake_parser.py", line 530, in _process_link_libraries
> assert len(args) >= 2
> AssertionError

I have just released a fix for this issue. I could not reproduce the other one, so I would need more information.

If you can, run with debug enabled (`haros --debug ...`) and paste the `log.txt` found in `~/.haros`.

droter, 2018-12-17:
Hi Andre,

Here are the results for this package: https://github.com/ros2/orocos_kinematics_dynamics.git

Terminal output:
`
[HAROS] Reading project and indexing source code...
Traceback (most recent call last):
  File "/usr/local/bin/haros", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 725, in main
    if launcher.launch(argv = argv):
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 150, in launch
    return args.command(args, settings)
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 184, in command_analyse
    return analyse.run()
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 497, in run
    configs, env = self._extract_metamodel(node_cache)
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 518, in _extract_metamodel
    extractor.index_source(settings = self.settings)
  File "/usr/local/lib/python2.7/dist-packages/haros/extractor.py", line 103, in index_source
    self._find_nodes(settings)
  File "/usr/local/lib/python2.7/dist-packages/haros/extractor.py", line 229, in _find_nodes
    extractor.find_nodes(pkg)
  File "/usr/local/lib/python2.7/dist-packages/haros/extractor.py", line 689, in find_nodes
    parser.parse(os.path.join(self.package.path, "CMakeLists.txt"))
  File "/usr/local/lib/python2.7/dist-packages/haros/cmake_parser.py", line 406, in parse
    self._analyse_command(command, args)
  File "/usr/local/lib/python2.7/dist-packages/haros/cmake_parser.py", line 435, in _analyse_command
    args = [self._argument(arg) for arg in args]
  File "/usr/local/lib/python2.7/dist-packages/haros/cmake_parser.py", line 778, in _argument
    return arg.replace('"', "")
AttributeError: 'list' object has no attribute 'replace'
`

debug log.txt output:

`
INFO:root:Executing selected command.
DEBUG:haros.data:HarosDatabase.load_definitions(/usr/local/lib/python2.7/dist-packages/haros/definitions.yaml)
DEBUG:haros.data:HarosDatabase.register rule max_coupling_objects_5
DEBUG:haros.data:HarosDatabase.register rule max_static_path_80
DEBUG:haros.data:HarosDatabase.register rule max_comment_ratio_30
DEBUG:haros.data:HarosDatabase.register rule no_global_names
DEBUG:haros.data:HarosDatabase.register rule mi_below_20
DEBUG:haros.data:HarosDatabase.register rule max_control_nesting_5
DEBUG:haros.data:HarosDatabase.register rule max_control_nesting_4
DEBUG:haros.data:HarosDatabase.register rule max_static_path_250
DEBUG:haros.data:HarosDatabase.register rule max_function_length_40
DEBUG:haros.data:HarosDatabase.register rule max_cyclomatic_complexity_10
DEBUG:haros.data:HarosDatabase.register rule mi_below_65
DEBUG:haros.data:HarosDatabase.register rule max_eloc_70
DEBUG:haros.data:HarosDatabase.register rule max_cyclomatic_complexity_15
DEBUG:haros.data:HarosDatabase.register rule max_weighted_methods_100
DEBUG:haros.data:HarosDatabase.register rule max_eloc_50
DEBUG:haros.data:HarosDatabase.register rule max_function_calls_10
DEBUG:haros.data:HarosDatabase.register rule max_comment_ratio_40
DEBUG:haros.data:HarosDatabase.register rule halstead_bugs_above_2
DEBUG:haros.data:HarosDatabase.register rule max_immediate_children_10
DEBUG:haros.data:HarosDatabase.register rule min_comment_ratio_20
DEBUG:haros.data:HarosDatabase.register rule max_weighted_methods_50
DEBUG:haros.data:HarosDatabase.register rule max_methods_available_20
DEBUG:haros.data:HarosDatabase.register rule max_function_calls_7
DEBUG:haros.data:HarosDatabase.register rule max_file_length_400
DEBUG:haros.data:HarosDatabase.register rule halstead_volume_above_8000
DEBUG:haros.data:HarosDatabase.register rule max_function_parameters_6
DEBUG:haros.data:HarosDatabase.register rule max_static_path_200
DEBUG:haros.data:HarosDatabase.register rule max_deepest_inheritance_5
DEBUG:haros.data:HarosDatabase.register metric cyclomatic_complexity
DEBUG:haros.data:HarosDatabase.register metric nested_control
[... 426 lines cut ...]
DEBUG:haros.extractor:Found file README.Debian at debian
DEBUG:haros.extractor:Found file rules at debian
DEBUG:haros.extractor:Found file liborocos-kdl.install at debian
DEBUG:haros.extractor:Found file substvars at debian
DEBUG:haros.extractor:PackageExtractor.populate(package:python_orocos_kdl)
INFO:haros.extractor:Indexing source files for package python_orocos_kdl
DEBUG:haros.extractor:Found file mainpage.dox at 
DEBUG:haros.extractor:Found file CMakeLists.txt at 
DEBUG:haros.extractor:Found file package.xml at 
DEBUG:haros.extractor:Found file manifest.xml at 
DEBUG:haros.extractor:Found file AMENT_IGNORE at 
DEBUG:haros.extractor:Found file rosdoc.yaml at 
DEBUG:haros.extractor:Found file PyKDLtest.py at tests
DEBUG:haros.extractor:Found file kinfamtest.py at tests
DEBUG:haros.extractor:Found file framestest.py at tests
DEBUG:haros.extractor:Found file frameveltest.py at tests
DEBUG:haros.extractor:Found file dynamics.sip at PyKDL
DEBUG:haros.extractor:Found file std_string.sip at PyKDL
DEBUG:haros.extractor:Found file framevel.sip at PyKDL
DEBUG:haros.extractor:Found file frames.sip at PyKDL
DEBUG:haros.extractor:Found file kinfam.sip at PyKDL
DEBUG:haros.extractor:Found file PyKDL.sip at PyKDL
DEBUG:haros.extractor:Importing cached Nodes.
DEBUG:haros.extractor:NodeExtractor.find_nodes(package:orocos_kdl)
`

git-afsantos, 2018-12-17:
I was able to fix the initial error, but now another error comes up in one of the secondary CMake files.

I cannot quick fix this one; the parser has to change significantly.
Since the current implementation is outdated and other users also had issues with it, I will prioritise changing the CMake parser more in future tasks.

closed 2018-12-17 by git-afsantos