git-afsantos/haros #33: haros.cmake_parser.IncompleteStatementError when runing haros on some packages

aleksandergabriel, 2018-12-17:
After trying HAROS on some simple packages containing mostly message and service definitions I tried to run it on a bit larger package but got this crash: 
```
haros full -p haros.yaml                                                                              
[HAROS] Loading common definitions...
[HAROS] Loading plugins...
  > Loaded mi_calculator
  > Loaded lizard_plugin
  > Loaded cppcheck_plugin
  > Loaded cpplint_plugin
  > Loaded pylint_plugin
  > Loaded cccc_plugin
  > Loaded ccd_plugin
  > Loaded radon_plugin
[HAROS] Reading project and indexing source code...
Traceback (most recent call last):
  File "/usr/local/bin/haros", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 725, in main
    if launcher.launch(argv = argv):
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 150, in launch
    return args.command(args, settings)
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 167, in command_full
    return (self.command_analyse(args, settings)
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 184, in command_analyse
    return analyse.run()
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 497, in run
    configs, env = self._extract_metamodel(node_cache)
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 518, in _extract_metamodel
    extractor.index_source(settings = self.settings)
  File "/usr/local/lib/python2.7/dist-packages/haros/extractor.py", line 103, in index_source
    self._find_nodes(settings)
  File "/usr/local/lib/python2.7/dist-packages/haros/extractor.py", line 229, in _find_nodes
    extractor.find_nodes(pkg)
  File "/usr/local/lib/python2.7/dist-packages/haros/extractor.py", line 689, in find_nodes
    parser.parse(os.path.join(self.package.path, "CMakeLists.txt"))
  File "/usr/local/lib/python2.7/dist-packages/haros/cmake_parser.py", line 398, in parse
    self.parser.parse(cmakelists)
  File "/usr/local/lib/python2.7/dist-packages/haros/cmake_parser.py", line 249, in parse
    self.parsetree = self.parse_block_children(None)
  File "/usr/local/lib/python2.7/dist-packages/haros/cmake_parser.py", line 271, in parse_block_children
    raise IncompleteStatementError()
haros.cmake_parser.IncompleteStatementError
```
I am new to HAROS so unfortunately I have no idea weather this is caused by HAROS or my code. As my code compiles I dont really know where to look for error.

git-afsantos, 2018-12-17:
Some part of your `CMakeLists.txt` got HAROS confused. This is likely a bug in the CMake parser HAROS is using, and not in your code. I have had a couple of issues with it before. It is outdated and in need of a replacement.

I can look into the issue and suggest a quick fix for it, but I need some more information.
Namely, I need to see which statement in the `CMakeLists.txt` is confusing the parser.

Is this package public? If so, can you share it?
If it is not public, let me know so that we can figure out a workaround.

gavanderhoorn, 2018-12-17:
[catkin_lint](https://github.com/fkie/catkin_lint) contains a CMake parser ([here](https://github.com/fkie/catkin_lint/blob/607a11aebd4a190d6847a4c7982664b67ef0f651/src/catkin_lint/cmake.py)). Might be worth a look?

git-afsantos, 2018-12-17:
Definitely. Another alternative I had noted down at the time was https://github.com/polysquare/cmake-ast.
Both seem to have received commits recently, so it might just be a matter of selecting the simplest one to use.

aleksandergabriel, 2018-12-17:
Package is private, but I see no reason which would block me to share CMakeLists with you. However I would still prefer to share it via some other private channel. 
However this is not really urgent and if you are planning to change parser in near future we can wait and then help with testing if needed.