git-afsantos/haros #39: Requirements file (missing bonsai.model)?

gavanderhoorn, 2019-03-03:
Trying to be a stupid user I only cloned the repository and then tried to use the "no install" variant (so tried `python haros-runner.py`).

On a "fresh" Ubuntu Xenial system this resulted in:

```python-traceback
Traceback (most recent call last):
  File "./haros-runner.py", line 24, in <module>
    from haros.haros import main
  File "/home/user/ros/haros/haros/haros.py", line 102, in <module>
    from .extractor import ProjectExtractor
  File "/home/user/ros/haros/haros/extractor.py", line 34, in <module>
    from bonsai.model import CodeGlobalScope, pretty_str
ImportError: No module named bonsai.model
```

Would it perhaps be an idea to provide a `requirements.txt` with the basic dependencies listed (yaml, bonsai, etc)? For the non-install option this would help avoid a nr of `ImportError`s.

git-afsantos, 2019-03-03:
Yes, that is a good idea :+1:

git-afsantos, 2019-03-04:
https://github.com/git-afsantos/haros/pull/44 should solve this.
This is an auto-generated file, using `pipreqs`, so it may not be 100% optimal/correct.

commit f31174c4 (2019-03-05): Merge pull request #44 from git-afsantos/dev-requirements-txt

Added requirements.txt, closes #39
closed 2019-03-05 by git-afsantos