git-afsantos/haros #40: pip install 'init' fails with missing rosgraph & pyflwor

gavanderhoorn, 2019-03-03:
Connected to #38.

`$me` being a stupid user again.

After a `pip install haros`, invoking `haros init` results in:

```python-traceback
Traceback (most recent call last):
  File "/home/user/venv_haros/bin/haros", line 6, in <module>
    from haros.haros import main
  File "/home/user/venv_haros/local/lib/python2.7/site-packages/haros/haros.py", line 102, in <module>
    from .extractor import ProjectExtractor
  File "/home/user/venv_haros/local/lib/python2.7/site-packages/haros/extractor.py", line 47, in <module>
    from .launch_parser import LaunchParser, LaunchParserError
  File "/home/user/venv_haros/local/lib/python2.7/site-packages/haros/launch_parser.py", line 30, in <module>
    import rosgraph.names
ImportError: No module named rosgraph.names
```

Running `source /opt/ros/kinetic/setup.bash` fixes that.

Another invocation then fails with:

```python-traceback
Traceback (most recent call last):
  File "/home/user/venv_haros/bin/haros", line 6, in <module>
    from haros.haros import main
  File "/home/user/venv_haros/local/lib/python2.7/site-packages/haros/haros.py", line 105, in <module>
    from .analysis_manager import AnalysisManager
  File "/home/user/venv_haros/local/lib/python2.7/site-packages/haros/analysis_manager.py", line 32, in <module>
    import pyflwor
ImportError: No module named pyflwor
```

I realise the ROS dependency is documented (implicitly) with:

> This assumes that you already have a working installation of ROS. 

Not every ROS user will have their shell automatically `source /opt/ros/kinetic/setup.bash` though, so perhaps that should be added to the readme.

Also: I realise the `pyflwor` dependency is actually documented in the *Requirements* section, but the command that installs it -- and the other dependencies -- don't appear in the readme until after we've been told to run `haros init`.

Perhaps the order of these parts of the readme should be reversed?

gavanderhoorn, 2019-03-03:
Any `haros` verb and `haros` itself actually fail with `ImportError: No module named rosgraph.names`.

git-afsantos, 2019-03-03:
Yes, there are a few assumptions that are not really well documented, as I have not had enough users yet to realise these oversights. One of them is the `source`ing of `setup.bash`.

I would like a more in depth fix for the README, possibly splitting it into multiple files, and document each of them a little bit better. In the meantime, I should take your suggestion of changing the order of those statements, so that things are less confusing.

git-afsantos, 2019-03-03:
The import error on all verbs is also expected, since I am not doing conditional imports (although I have been thinking about this as a way to solve a few issues, and possibly reduce dependencies).

git-afsantos, 2019-03-05:
Both #43 and #44 should temporarily solve this, right? Minus the overall improvements on the README.

gavanderhoorn, 2019-03-05:
I've commented on #43 and #44.

git-afsantos, 2019-03-09:
The README was updated with a note on the requirements of `setup.bash`, see https://github.com/git-afsantos/haros/commit/ab05399e7622fb7adfe024e92cab46067d8c7c44

I think I have minimally addressed all points in this issue, until I get to work on a proper rehaul of the documentation, so I will be closing this. Reopen if you feel something should be better addressed.

closed 2019-03-09 by git-afsantos