git-afsantos/haros #41: 'haros init' fails with "KeyError: ROS_WORKSPACE"

gavanderhoorn, 2019-03-03:
After resolving #40 (so with the base `/opt/ros/kinetic/setup.bash` sourced and `pyflwor` installed), running `haros init` results in:

```python-traceback
[HAROS] It seems this is a first run.
Traceback (most recent call last):
  File "/home/user/venv_haros/bin/haros", line 10, in <module>
    sys.exit(main())
  File "/home/user/venv_haros/local/lib/python2.7/site-packages/haros/haros.py", line 725, in main
    if launcher.launch(argv = argv):
  File "/home/user/venv_haros/local/lib/python2.7/site-packages/haros/haros.py", line 142, in launch
    self.command_init(args, HarosSettings())
  File "/home/user/venv_haros/local/lib/python2.7/site-packages/haros/data.py", line 454, in __init__
    self.workspace = workspace or self.find_workspace()
  File "/home/user/venv_haros/local/lib/python2.7/site-packages/haros/data.py", line 504, in find_workspace
    raise KeyError("ROS_WORKSPACE")
KeyError: 'ROS_WORKSPACE'
```

Only after having `source`d a `setup.bash` of a ROS workspace does `init` continue and clones the plugins repository.

git-afsantos, 2019-03-03:
Indeed, the current version more or less assumes that you already have a working ROS development environment. Perhaps I should try to relax that behaviour a bit, as I am getting more use cases that do not follow that pattern.

gavanderhoorn, 2019-03-04:
> Indeed, the current version more or less assumes that you already have a working ROS development environment.

So not having a local workspace doesn't fulfil that requirement?

git-afsantos, 2019-03-05:
The original idea was that HAROS would be used to analyse the source of your own projects, and so we could make the assumption that you would have a workspace, and that it would be `source`d. In most setups, I do not think `/opt/ros` would have any source code (C++ at least, besides headers), and that is why I dismissed that case.

Do you think this is too strict? I could look into what I would have to change to relax this requirement.

gavanderhoorn, 2019-03-05:
> The original idea was that HAROS would be used to analyse the source of your own projects, and so we could make the assumption that you would have a workspace, and that it would be `source`d. In [...]

Makes sense.

> Do you think this is too strict? I could look into what I would have to change to relax this requirement.

It's not necessarily too strict, I just wanted to make sure I understood things correctly.

I don't think it makes much sense to be able to scan `/opt/ros` as it only includes binaries.

Although it does contain Python for all the Python nodes. Those could still be parsed -- if HAROS would support Python.

gavanderhoorn, 2019-03-05:
I would recommend printing a more user friendly error message though than a `KeyError` ;)

git-afsantos, 2019-03-05:
> Although it does contain Python for all the Python nodes. Those could still be parsed -- if HAROS would support Python.

Andrzej has a student working on Python support, as far as I know.

Parsing those nodes should not be a problem, I think. HAROS more or less replicates `rospack find <pkg>`, and then goes from wherever the package is.

> I would recommend printing a more user friendly error message though than a `KeyError` ;)

There are quite a lot of error messages that need improvement, indeed! I should make a list of known cases and open an issue for that.

closed 2019-03-09 by git-afsantos

git-afsantos, 2019-03-09:
@gavanderhoorn I have remedied the `KeyError` issue with a nicer message. Further work on error handling is tracked in #49.

commit 2691a116 (2019-03-09): Merge pull request #48 from git-afsantos/dev-issue-41

Provided better error message for when workspace is not present. Closes #41