git-afsantos/haros #52: Melodic Updates

JWhitleyAStuff, 2019-04-02:
Are there currently any plans to support 18.04/Melodic? If so, is there a timeline?

git-afsantos, 2019-04-02:
Could you be more specific? Is there something that is not working in Melodic? Something related to C++14, perhaps?

I can set up a machine with Melodic to try it out, but in principle most of the tools should work right away.

JWhitleyAStuff, 2019-04-02:
Mostly it's the version of clang. 3.8 is not available in Ubuntu 18.04.

gavanderhoorn, 2019-04-03:
re: clang: see #42.

JWhitleyAStuff, 2019-04-08:
Thanks. I'll continue discussion there.

closed 2019-04-08 by JWhitleyAStuff

git-afsantos, 2019-04-13:
@JWhitleyAStuff Sorry for the late response. I have just posted an update on the other issue (https://github.com/git-afsantos/haros/issues/42#issuecomment-482895476).