git-afsantos/haros #62: Plugin

nlimpert, 2019-06-03:
During the initial setup operations `haros` executes `git` to either clone or update the plugins repository found at https://github.com/git-afsantos/haros_plugins:

https://github.com/git-afsantos/haros/blob/45b4bbd0644e4b1edbf4177531ba2b863a3db211/haros/haros.py#L406-L419

As this obviously requires `git` to be installed this causes an issue when using `haros` from within [`haros_catkin`](https://github.com/rosin-project/haros_catkin) inside a buildfarm docker container. The docker container of interest is the `docker_build_and_test` which won't include git by default (cf. https://github.com/rosin-project/haros_catkin/issues/10).

Would it be better to release the plugins in a separate package via `pip` instead of manually invoking `git`?

nlimpert, 2019-06-03:
Thanks again! Much appreciated!

I was hesitating to extend https://github.com/git-afsantos/haros/issues/14 with this but I think it is still related.

git-afsantos, 2019-06-03:
I just released `3.3.0` with #63, which should solve this issue.

closed 2019-06-05 by git-afsantos