git-afsantos/haros #64: Should rosgraph be added to requirements.txt?

gavanderhoorn, 2019-06-04:
That package will probably always be installed when users have ROS installed, but if `requirements.txt` is supposed to capture the Python modules necessary to be able to run haros, then it would seem that `rosgraph` should be listed.

On systems with ROS installed, `pip` will probably detect it and decide that the dependency has already been satisfied. On systems without, it will be installed.

gavanderhoorn, 2019-06-04:
It's used here:

https://github.com/git-afsantos/haros/blob/c2ecbfa432fafcc7adbde543e205439be00f8d2e/haros/launch_parser.py#L30

git-afsantos, 2019-06-04:
Makes sense to me. Thanks for the PR! :+1:

gavanderhoorn, 2019-06-04:
It may even be that `rosgraph` is not released on `pypi` :(

I should've checked better (doing too many things at the same time right now).

git-afsantos, 2019-06-04:
I cannot find it. Unless it is bundled within some other distribution, I think it may not be released individually on PyPI.

gavanderhoorn, 2019-06-04:
Just checked: it's a ROS package, not a pure Python one.

Submitted #66.

Sorry for the noise and the work.

gavanderhoorn, 2019-06-04:
It would still be good to have some way of expressing this dependency though.

Perhaps a `try: .. except: ..` around the import would be an idea? That way we at least can print a more user-friendly error message instead of an import error.

git-afsantos, 2019-06-04:
> Perhaps a `try: .. except: ..` around the import would be an idea? That way we at least can print a more user-friendly error message instead of an import error.

I was thinking about this again. I will make it a lazy import, and provide a warning and a fallback, in case it is missing.

closed 2019-06-04 by git-afsantos

gavanderhoorn, 2019-06-04:
The fallback is perhaps not necessary, and my use-case is slightly exotic.

Even just a nicer error would be great.

git-afsantos, 2019-06-04:
Addressed and fixed in #67 and #68 @gavanderhoorn

commit 07c62275 (2019-06-04): requirements: add rosgraph. Fix #64.

Used by `launch_parser.py`, so should be listed.
commit 91d2764e (2019-06-04): Merge pull request #65 from gavanderhoorn/patch-1

requirements: add rosgraph. Fix #64.
commit a7492a4b (2019-06-04): Merge pull request #66 from gavanderhoorn/revert-65-patch-1

Revert "requirements: add rosgraph. Fix #64."
commit d749e52e (2019-06-04): Revert "requirements: add rosgraph. Fix #64."