git-afsantos/haros #72: Plugins not found

max-krichenbauer, 2019-08-08:
Hi!

I installed HAROS via pip, and also installed haros_plugins via pip.
I also followed the comment in [issue 20](https://github.com/git-afsantos/haros/issues/20) and downloaded the plugins repo to ```~/.haros/plugins/```

but when I try using the ```-w``` flag, it prints:
```
[HAROS] Running setup operations...
[HAROS] Loading common definitions...
[HAROS] Loading plug-ins...
ERROR:root:There are no analysis plugins.
```

git-afsantos, 2019-08-08:
Hello!

I suspect that you just provided an invalid plugin name for the `-w` option.
I just did a fresh install to test this, and I cannot reproduce your problem otherwise.

I did

```
pip install haros_plugins
pip install haros
haros analyse -w haros_plugin_mi_calculator -p project.yaml
```

Which runs fine. However, if I introduce an unknown plugin name, I get the same error as you do.

```
haros analyse -w mi_calculator -p project.yaml
```

Can you paste here the commands that you tried?

---

Also, if your Haros version is newer than `3.3.0`, you should not need the `~/.haros/plugins/` directory anymore, `pip install haros_plugins` is enough.

I also see that I have not updated the README, so I will be doing it now.

max-krichenbauer, 2019-08-09:
Thank you for the quick reply!
You were right - naming was the issue. The documentation did not mention that the plugin name must be precedet by "haros_plugin_", and the error message made it sound like no plugin (at all) were found instead of just this particular plugin not being found.

Now I can activate using cppcheck, but I'm a bit confused by the issues it finds when using it through HAROS:

Even if I have blatant pointer mishandling in my code, they do not show up in the HAROS output.

If I run `cppcheck test.cpp`, it does find issues:
> Checking test.cpp...
> [test.cpp:89]: (error) Memory leak: bad_var
> [test.cpp:12]: (error) Uninitialized variable: bad_var

But when I run `haros full -w haros_plugin_cppcheck -p myproject.yaml` I only get:

> Rule No Redundant Variables
> Ensure that no expression is redundant.
> Variable 'bad_var' is assigned a value that is never used.

Does HAROS run cppcheck with special parameters that disable the static analysis in favor of mere code style analysis?

max-krichenbauer, 2019-08-09:
Looking at the plugins source code, I found the answer: [haros_plugin_cppcheck/rules.xml](https://github.com/git-afsantos/haros_plugins/blob/master/haros_plugin_cppcheck/rules.xml) makes cppcheck only look for 26 specific source code style rule violations.

What's the reasoning behind this?
Is it planned to extend this scope in the future?

git-afsantos, 2019-08-09:
> The documentation did not mention that the plugin name must be preceded by "haros_plugin_"

Good point. It probably should not require the prefix, since it is an implementation detail to find plugins in the system.

> the error message made it sound like no plugins (at all) were found instead of just this particular plugin not being found.

I already have an open issue to improve error reporting. I will add this to the list.

> [...] makes cppcheck only look for 26 specific source code style rule violations. What's the reasoning behind this?

None in particular. I just have not really worked on that plugin since my MSc thesis (4 years ago) which was mostly about coding style violations. Also, I was targeting Ubuntu Trusty / ROS Indigo at the time, and CppCheck has improved a lot since.

> Is it planned to extend this scope in the future?

It only makes sense to do so, since CppCheck is free and detects a number of useful things.

---

I will open the respective issues for each of your points and try to address them as soon as possible.

git-afsantos, 2019-08-09:
I just fixed the `-w` issue and slightly improved the error messages. Providing a plugin name without the prefix should work now, e.g.:

```
haros analyse -w cppcheck -p myproject.yaml
```

Updating the CppCheck plugin requires a bit more work, so it is still *to do* for now.

git-afsantos, 2019-08-09:
I have just pushed a quickfix for the CppCheck plugin. It now reports all errors, but the ones that are not in the original list do not have as nice an error report.

They look something like this:

> **CppCheck Check**
> **Error reported by CppCheck.**
> *Actual error message reported by CppCheck*

I will open an issue to improve this error reporting later on.

The updated versions are already released on `pip`, both for `haros` and `haros_plugins`.

max-krichenbauer, 2019-08-10:
Great! Thank you!

closed 2019-08-10 by max-krichenbauer