git-afsantos/haros #92: Exclude subdirectories by pattern

StefanFabian, 2019-11-23:
Just running `haros analyze` will also find issues in folders generated by IDEs not related to the actual application. 
E.g., CLion will create a cmake-build-debug folder in which haros finds a staggering amount of issues.
Is there an option to ignore these folders, e.g., using a wildcard ignore approach?

git-afsantos, 2019-11-23:
I do not think the current version supports this. I could add it to the next version.

Meanwhile, where is this directory located? Can you provide more details regarding how you are calling Haros, and what the directory structure looks like?

StefanFabian, 2019-11-23:
I just used `haros analyze` at the root of my workspace (in the src folder).
The folder structure is usually `src/{repo}/{package}/(cmake-build-{debug,release}/...)`.
The last one only exists if the package was ever opened with CLion which runs a build process for an improved autocompletion.

git-afsantos, 2019-11-23:
I see. In that case, I think there is no feasible workaround.

I will see if I can sneak that feature in for the next release, and get back to you soon.

git-afsantos, 2019-11-24:
I just updated Haros to solve this issue @StefanFabian. Please update to [version 3.8](https://github.com/git-afsantos/haros/releases/tag/v3.8.0) and give it a try.

The solution:

Go to your configurations file (defaults to `~/.haros/configs.yaml`) and add an `analysis:ignore:files` list to it. In this list you can add glob patterns that will match paths of the form `{pkg}/path/to/{file}`.

For your example above:

```yaml
analysis:
  ignore:
    files:
      - "cmake-build-debug/*"
      - "cmake-build-release/*"
```

Everything under these two directories should be ignored for all packages.

Let me know how it goes. I will reopen the issue if this does not work for you.

StefanFabian, 2019-11-25:
Can't check if it works now because running `haros init` now fails with:
```
Traceback (most recent call last):
  File "/home/stefan/.local/bin/haros", line 8, in <module>
    sys.exit(main())
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/haros.py", line 904, in main
    if launcher.launch(argv = argv):
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/haros.py", line 170, in launch
    self.minimal_output = args.minimal_output
AttributeError: 'Namespace' object has no attribute 'minimal_output'
```

and `haros analyse` fails with:
```
[HAROS] Running setup operations...
[HAROS] Loading common definitions...
[HAROS] Loading plugins...
  > Loaded haros_plugin_cccc
  > Loaded haros_plugin_ccd
  > Loaded haros_plugin_cppcheck
  > Loaded haros_plugin_cpplint
  > Loaded haros_plugin_lizard
  > Loaded haros_plugin_mi_calculator
  > Loaded haros_plugin_pylint
  > Loaded haros_plugin_radon
[HAROS] Reading project and indexing source code...
WARNING:haros.extractor:Parsing error in /home/stefan/hector/src/hector_vision/hector_motion_detection/launch/test_with_bagfiles.launch:
'not well-formed (invalid token): line 5, column 1'
WARNING:haros.extractor:Parsing error in /home/stefan/hector/src/perception_pcl/pcl_ros/samples/pcl_ros/surface/sample_convex_hull.launch:
'missing required attribute: file'
WARNING:haros.extractor:C++ AST parser not found.
WARNING:haros.cmake_parser:Found 'include_directories' in CMake with no arguments, but expected at least one.
WARNING:haros.cmake_parser:Found 'include_directories' in CMake with no arguments, but expected at least one.
Traceback (most recent call last):
  File "/home/stefan/.local/bin/haros", line 8, in <module>
    sys.exit(main())
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/haros.py", line 904, in main
    if launcher.launch(argv = argv):
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/haros.py", line 183, in launch
    return args.command(args)
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/haros.py", line 220, in command_analyse
    return analyse.run()
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/haros.py", line 596, in run
    configs, nodes, env = self._extract_metamodel(node_cache, rules)
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/haros.py", line 619, in _extract_metamodel
    extractor.index_source(settings = self.settings)
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/extractor.py", line 165, in index_source
    self._find_nodes(settings)
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/extractor.py", line 320, in _find_nodes
    extractor.find_nodes(pkg)
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/extractor.py", line 920, in find_nodes
    parser.parse(cmake_path)
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/cmake_parser.py", line 424, in parse
    self._analyse_control_flow(command, args, children)
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/cmake_parser.py", line 452, in _analyse_control_flow
    self._analyse_command(child_command, child_args)
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/cmake_parser.py", line 484, in _analyse_command
    self._process_install(args)
  File "/home/stefan/.local/lib/python2.7/site-packages/haros/cmake_parser.py", line 525, in _process_install
    for file in os.listdir(os.path.join(self.directory, dir))
OSError: [Errno 2] No such file or directory: '/include/eigen3'
```

git-afsantos, 2019-11-25:
I just released version 3.8.1 which should fix both issues.

Since a few versions back, `init` should not be needed anymore, under normal circumstances.

From the looks of the traceback, I think that you have a CMakeLists.txt that looks something like

```cmake
install(DIRECTORY /include/eigen3 ...)
```

Is this the case? Does it work with catkin?

StefanFabian, 2019-11-30:
No, the only thing I could find using grep was:
```
set(EIGEN_CATKIN_EIGEN_INCLUDE ${CATKIN_DEVEL_PREFIX}/include/eigen3)
```

The workspace works fine using catkin build.

---
Sorry for the late reply, looks like I forgot to press 'Comment' ^^

Excluding the folders seems to work now. Thank you!

closed 2019-11-30 by StefanFabian