git-afsantos/haros #94: ROS Communications Section Does Not Populate

RoyallDesigns, 2020-02-12:
When attempting to run `haros` with the `full` sub-command and with the `--project-file` argument pointing to a custom project file, the HTML output does not display any summary information for the `ROS Communications` Section on the `Dashboard` page; only `n/a` appears for the `Topics`, `Remappings`, etc. I have taken the proper steps to install all required Python / Apt dependencies, have built the ROS Packages specified in my project file with Clang as the compiler, and am using the `--parse-nodes` flag when running with the `full` sub-command. The `Models` tab does display with the appropriate node names, parameter names, topic names, etc. but the high-level summary of items like the number of topics never appears in the `ROS Communications` Section.

I am using the latest commit of `haros` (commit `1cd4774dd6863cab9edefd52dda3c47fadbaea7e`) with ROS Melodic on Ubuntu 18.04. Any information about whether this is a known issue and how to remedy it or whether there are telltale signs to help identify the cause of this issue would be extremely helpful.

git-afsantos, 2020-02-12:
Thanks for the detailed report!
If I remember correctly, I never got to implementing that because I could not figure out what to populate that section with. The section was designed before the implementation of model extraction.

The issue is that a single project can have multiple different configurations (for instance, one for every launch file). A sum of all the numbers would be useless to print (many nodes and topics might repeat from one configuration to another). Do you have any suggestion? Perhaps an average number? It would work for single configurations, and would still have some use with multiple configurations.

Let me know also if you would rather see something else entirely in that place. Maybe I could move that section over to the graph visualization page, where a single configuration is already selected and the numbers make sense.

RoyallDesigns, 2020-02-13:
I personally believe that moving the `ROS Communications` section over to the graph visualization page would be more organized and would provide more meaning to this summary / these stats. This would be more useful when combined with the idea that this section would be populated based on the currently selected configuration. This way, there is no notion of `ROS Communications` stats for the combined collection of project configurations. 

How open are you to future input and possible PRs on this project?

git-afsantos, 2020-02-13:
Yes, I agree. I will move that panel over to the models' page.

> How open are you to future input and possible PRs on this project?

Definitely open. The next couple of months are quite busy for me though, so I might be slower than usual to implement/fix stuff.

closed 2020-02-19 by git-afsantos

git-afsantos, 2020-02-19:
@RoyallDesigns I changed the Dashboard and Models pages in [version 3.9](https://github.com/git-afsantos/haros/releases/tag/v3.9.0). Feel free to try it out and let me know if there are any further issues.

PR #95 (2020-02-19): Fix issue #94
fixes #94