git-afsantos/haros #97: Catkin_tools compatibility

AlexisTM, 2020-04-02:
`catkin_tools` works slightly differently from `catkin_make` but is also one of the most used tools as it proposes Toolchains, profiles and workspace extension (the main features we use).

How are we supposed to configure HAROS for catkin_tools? How `ROS_WORKSPACE` (unset) supposed to be? How can we handle the face we have the source and build folder apart from each other (depending on catkin_tools config).

git-afsantos, 2020-04-02:
Indeed, HAROS has not received built-in support for `catkin_tools` yet, but I think you can work around it. I assume you want to try model extraction, and thus need the workspace and build location.

In the HAROS home directory (defaults to `~/.haros`), you will find a `configs.yaml` file.
In this file, you can un-comment and edit a few lines to point to the right places.

```yaml
workspace: '/path/to/ws'
cpp:
    compile_db: '/path/to/ws/build'
```

I am not sure if the CMake parser will give you trouble still. Let me know.

AlexisTM, 2020-04-03:
https://github.com/rosin-project/haros_catkin/tree/735a6c6316d9b72754085bc30739accad673354f

AlexisTM, 2020-04-06:
When doing so, it finds the packages in `/tmp/src/install_x64` instead of `/home/alexis/workspace/src` (The install folder that has been sourced)

```yaml
%YAML 1.1
---
workspace: "/home/alexis/workspace/src"
cpp:
    parser_lib: '/usr/lib/llvm-11/lib'
    std_includes: '/usr/lib/llvm-11/lib/clang/11.0.0/include'
    compile_db: '/tmp/src/build_x64'
py:
    compile_db: '/tmp/src/build_x64'
packages:
  - base
  - ncc
```

git-afsantos, 2020-04-07:
Ok, then I will have to look further into it. I probably have to change something in the function that searches for packages.

By the way, the `packages` section does not go into the `configs.yaml`, it goes into a separate YAML file (unless you are passing `configs.yaml` as a `-p` argument).

Meanwhile, if you want to try something else, try running in the source directory or passing the directory as `-c`, e.g.:

```bash
haros -c /home/alexis/workspace/src analyse ...
```

I think this will prioritise this directory in the search paths.

closed 2024-06-24 by git-afsantos