git-afsantos/haros #99: Can't get the topics in "Models"

LeiGong-USTC, 2020-04-10:
I  just followed the tutorial and used haros to analyse the haros_tutorials, but when i runs the visualisation i can not get any topics in the Models  

[image]
but the demo you show on the https://git-afsantos.github.io/haros/#dashboard can ,so i just don't know what's the problem is  
[image]

git-afsantos, 2020-04-10:
What commands did you run?
Did you follow the steps in [Model Extraction](https://github.com/git-afsantos/haros/blob/master/MODEL_EXTRACTION.md)?

In particular, did you compile the code with clang *before* running HAROS, and did you generate a `compile_commands.json` file?

LeiGong-USTC, 2020-04-10:
the command  is `python haros-runner.py full  -n -p  /path/to/my_index.yaml`
i just use the catkin_make to build and source the devel/setup.bash,but i didn't compile the code with clang,maybe that's the reason ,i will try this ,thanks for your help !

LeiGong-USTC, 2020-04-11:
I have solved the problem
i compiled the code with clang  but it still didn't work , then i change the `workspace` and `cpp:compile_db`  in ~/.haros/configs.yaml  ,make sure that they are pointing to my workspace directory,however it still didn't work.
finally , i deleted the old haros_tutorials project  in ~/.haros/projects ,  then run the command `python haros-runner.py full -n -p /path/to/my_index.yaml` and the problem solved

git-afsantos, 2020-04-11:
Good to know! :+1:

closed 2020-04-11 by git-afsantos