git-afsantos/haros #101: After pip install, 'haros init' fails with 'ImportError: no module named haros'

evpitts, 2020-09-16:
I ran pip install haros on Ubuntu 16.04, which gave me

 `ERROR: Cannot uninstall 'enum34'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.`

So I tried again with the --ignore-installed flag and it ran without error. However, when I try a 'haros init', it fails with
`Traceback (most recent call last):
  File "/home/droge-robotics/.local/bin/haros", line 7, in <module>
    from haros.haros import main
ImportError: No module named 'haros'`

git-afsantos, 2020-09-17:
Seems like a problem in the Python environment, and not with Haros itself.

Is this a `virtualenv` or system install? Python 2.7 or 3.x?

git-afsantos, 2020-09-17:
I just tried a fresh install in a new virtual environment and everything worked without an issue.

```bash
$ virtualenv issue
$ cd issue/
$ source bin/activate
$ pip install haros
$ haros init
```

No errors found, got the expected output. This was on Ubuntu 18.04, not 16.04, but I doubt that is the root of the problem.

evpitts, 2020-09-21:
I tried it in both the system install and a virtual environment. This is 2.7

git-afsantos, 2020-09-21:
It works under Python 2.7, so that is not the issue.

As I said, I have tested it multiple times with fresh virtual machines, and never ran into an issue.
Something is different about your setup or environment.
Without further details (commands used to install, etc.), I am afraid I cannot help much.

Have you tried the steps I listed in the previous comment?

evpitts, 2020-09-21:
I have tried those steps. I'm still getting the same error. What further details would be helpful?

git-afsantos, 2020-09-22:
At this point I can only google around, same as you can.

I see that there are other projects where people ran into similar errors, and the solution seems to be reinstalling the faulty package (`enum34` in your case) via `pip`, instead of `apt-get`.

Do it at your own risk. Here are some links that may be helpful.

https://stackoverflow.com/questions/53807511/pip-cannot-uninstall-package-it-is-a-distutils-installed-project

https://askubuntu.com/questions/1050158/problem-installing-tensorflow

closed 2021-06-25 by git-afsantos