git-afsantos/haros #105: Trying to run Haros in our CI workflow but we run into:  AttributeError: 'list' object has no attribute 'get'

etienn8, 2021-03-02:
Hi,
We added Haros to our CI workflow few months ago, but for some reasons, yesterday, it stopped working. I re-ran our CI workflow  on a old Pull Request that ran Haros without a problem and it failed. Do you have any idea on your side why this is happening?

Thanks

Log: 

```
Run . /opt/ros/$ROS_DISTRO/setup.sh
  . /opt/ros/$ROS_DISTRO/setup.sh
  haros analyse -p src/SwarmUS-ROS/haros_config.yaml -b haros_plugin_radon haros_plugin_ccd haros_plugin_cppcheck haros_plugin_cpplint -n
  mkdir haros
  haros export -v ./haros
  shell: sh -e {0}
WARNING:haros.plugin_manager:Could not find blacklisted plugin: haros_plugin_radon
WARNING:haros.plugin_manager:Could not find blacklisted plugin: haros_plugin_ccd
WARNING:haros.plugin_manager:Could not find blacklisted plugin: haros_plugin_cppcheck
WARNING:haros.plugin_manager:Could not find blacklisted plugin: haros_plugin_cpplint
WARNING:haros.extractor:C++ AST parser not found.
WARNING:haros.cmake_parser:Found 'include_directories' in CMake with no arguments, but expected at least one.
Traceback (most recent call last):
[HAROS] Running setup operations...
  File "/usr/local/bin/haros", line 8, in <module>
[HAROS] Loading common definitions...
    sys.exit(main())
[HAROS] Loading plugins...
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 1002, in main
  > Loaded haros_plugin_cccc
  > Loaded haros_plugin_lizard
  > Loaded haros_plugin_mi_calculator
  > Loaded haros_plugin_pylint
[HAROS] Reading project and indexing source code...
  > Parsing nodes might take some time.
    if launcher.launch(argv = argv):
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 184, in launch
    return args.command(args)
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 225, in command_analyse
    return analyse.run()
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 631, in run
    self._extract_configurations(self.database.project, configs, nodes, env)
  File "/usr/local/lib/python2.7/dist-packages/haros/haros.py", line 703, in _extract_configurations
    cfg.user_attributes = data.get("user_data", {})
AttributeError: 'list' object has no attribute 'get'
Error: Process completed with exit code 1.
```

git-afsantos, 2021-03-03:
Hi @etienn8 ,

I think it might be a bug on my side. I will fix it shortly.
Just to make sure, can you confirm that your YAML file has somewhere a configuration that is simply a list of launch files?
Something like this:

```yaml
configurations:
  myconfig:
    - pkg/launch1.launch
    - pkg/launch2.launch
```

git-afsantos, 2021-03-03:
This particular bug should be fixed with commit 918813883b03138a06d0c822d23eb879284e5601.
Please update to version 3.9.2 ([GitHub](https://github.com/git-afsantos/haros/releases/tag/v3.9.2), [PyPI](https://pypi.org/project/haros/#files), `pip install -U haros`) and let me know.

etienn8, 2021-03-16:
Our problems are resolved, thanks!

closed 2021-03-16 by git-afsantos