git-afsantos/haros #111: FileNotFoundError: [Errno 2] No such file or directory: '/home/tim/.haros/log.txt'

Timple, 2021-08-19:
I received the error of an unexisting log file upon running haros:

```
FileNotFoundError: [Errno 2] No such file or directory: '/home/tim/.haros/log.txt'
```

Full log:

```shell
$ pip install --user haros --upgrade
Collecting haros
  Downloading haros-3.10.13.tar.gz (323 kB)
     |████████████████████████████████| 323 kB 6.0 MB/s 
Requirement already satisfied, skipping upgrade: bonsai-code<1.0.0,>=0.5.0 in ./.local/lib/python3.8/site-packages (from haros) (0.6.3)
--SKIPSTUFF--
Successfully installed haros-3.10.13

$ haros --debug analyse
Traceback (most recent call last):
  File "/home/tim/.local/bin/haros", line 8, in <module>
    sys.exit(main())
  File "/home/tim/.local/lib/python3.8/site-packages/haros/haros.py", line 1047, in main
    if launcher.launch(argv=argv):
  File "/home/tim/.local/lib/python3.8/site-packages/haros/haros.py", line 183, in launch
    logging.basicConfig(filename=self.log_path, filemode="w",
  File "/usr/lib/python3.8/logging/__init__.py", line 1988, in basicConfig
    h = FileHandler(filename, mode)
  File "/usr/lib/python3.8/logging/__init__.py", line 1147, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.8/logging/__init__.py", line 1176, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
FileNotFoundError: [Errno 2] No such file or directory: '/home/tim/.haros/log.txt'
```

git-afsantos, 2021-08-19:
Thanks for reporting the error.
At a first glance, it should not be too hard to solve. I will try to look into it in a few hours and will let you know once it is solved.

I just noticed that I had also registered the same report in #70.

git-afsantos, 2021-08-19:
I believe this should be fixed with `v3.10.14`.

commit df875b58 (2021-08-19): fixes #70
fixes #111
closed 2021-08-19 by git-afsantos