git-afsantos/haros #122: Model visualization is empty with no apparent errors

hsd-dev, 2024-06-17:
HAROS version: 3.10.20

Below are the steps I followed:

#### Install deps
```sh
sudo apt-get install -y cppcheck cccc python-is-python3 && pip3 install bloom clang==10.0.1 haros
```

#### define `project.yaml`
```yaml
%YAML 1.1
---
packages:
  - package_a
  - package_b
  - package_c
```

#### Init
```
haros init
```
#### /root/.haros/configs.yaml (running in a docker container)
```yaml
%YAML 1.1
---
workspace: "/workspace"
environment: null
plugin_blacklist: []
cpp:
  parser_lib: "/usr/lib/llvm-10/lib"
  std_includes: "/usr/lib/llvm-10/lib/clang/10.0.0/include"
  compile_db: "/workspace/build"
analysis:
  ignore:
    tags: []
    rules: []
    metrics: []
```
#### full run
```sh
haros full -p /haros.yaml
```
#### Output
```sh
[HAROS] Running setup operations...
[HAROS] Check news updates with:
  $ haros news
[HAROS] Running setup operations...
[HAROS] Loading common definitions...
[HAROS] Loading plugins...
  > Loaded haros_plugin_cccc
  > Loaded haros_plugin_ccd
  > Loaded haros_plugin_cppcheck
  > Loaded haros_plugin_cpplint
  > Loaded haros_plugin_lizard
  > Loaded haros_plugin_mi_calculator
  > Loaded haros_plugin_pylint
  > Loaded haros_plugin_radon
[HAROS] Reading project and indexing source code...
[HAROS] Running analysis...
WARNING:haros.analysis_manager:Could not import pyflwor. Skipping query execution.
ERROR:haros.analysis_manager:Plugin haros_plugin_radon ran into an error.
[HAROS] Saving analysis results...
[HAROS] Serving visualisation at localhost:8080
[HAROS] Press enter to shutdown the viz server:127.0.0.1 - - [17/Jun/2024 14:02:25] "GET / HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /lib/css/font-awesome.min.css HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /lib/css/pure-min.css HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /lib/css/grids-responsive-min.css HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /css/type.css HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /css/color.css HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /css/style.css HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /lib/js/jquery-2.2.1.min.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /lib/js/underscore.min.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /lib/js/backbone.min.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /lib/js/d3.min.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /lib/js/dagre.min.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /js/app.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /js/models.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /js/views/packages.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /js/views/issues.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /js/views/general.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /js/views/dashboard.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /js/views/ros.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /js/views/help.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /js/routes.js HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] code 404, message File not found
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /favicon.png HTTP/1.1" 404 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /data/projects.json HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /data/default/rules.json HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /data/default/packages.json HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /data/default/summary.json HTTP/1.1" 200 -
127.0.0.1 - - [17/Jun/2024 14:02:25] "GET /data/default/configurations.json HTTP/1.1" 200 -
```
The dashboard populates some issues including code metrics, but the model viz is empty.

git-afsantos, 2024-06-20:
Hello, Harsh!
I see that you installed the `clang` *Python bindings*, but did you install `libclang` too (of the same version)?

Since you did

```bash
pip3 install clang==10.0.1
```

Did you also run this (or something similar)?

```bash
sudo apt-get install libclang-10.0.1-dev
```

Basically, these paths have to exist, otherwise you are missing a dependency.

- `/usr/lib/llvm-10/lib`
- `/usr/lib/llvm-10/lib/clang/10.0.0/include`

git-afsantos, 2024-06-20:
I am just trying to understand if you followed all these steps:
https://github.com/git-afsantos/haros/blob/master/INSTALL.md#model-extraction

Before asking you for further debugging logs.

git-afsantos, 2024-06-20:
Also, you may find this repository useful:
https://github.com/git-afsantos/make-haros-easy

It already has Docker files with everything set up. Maybe you can adapt it for your use case, since you mentioned a Docker container.